atopile/atopile #106: `ato create` forces the creation of a repository

aleokdev, 2024-02-07:
Hello, I'm trying out atopile for the first time. I'm trying to integrate it into an already existing project, inside one of its folders. I tried using `ato create`, but it seems it doesn't allow skipping the repository creation.

It would be great if `ato create` asked if you wanted a repository or not, or even better, detected if the CWD already had one and skips the step entirely if it exists (kind of like what `cargo` does).

sdaglish, 2024-02-07:
I just ran into the same problem. I'm testing out creating a basic project, and don't want to create a GH repo just for a small test.

I get creating the local git repo, but it seems odd forcing the use of a GH repo.

mawildoer, 2024-02-07:
We currently have a prerelease of this working and will make a proper release for it today.

You can use ‘pip install atopile==0.2.7.dev0’ to get it now!

Thanks for the report 🙏

mawildoer, 2024-02-07:
Released in 0.2.7!

closed 2024-02-07 by mawildoer

aleokdev, 2024-02-07:
That was quick! Thanks! 😊

aleokdev, 2024-02-11:
It seems to still be creating a repository even if the option to create one is specified to off. Had to remove the .git folder, didn't realize it was there until I switched to repo view on VSCode...

mawildoer, 2024-02-11:
I can change that specific behaviour. I was considering whether to fully blow away the ‘.git’ directory or just stub it. Can potentially do the git repo detection, but for the minute copying out the content should work fine.
Definitely agreed that we shouldn’t force creating a new remote.

mawildoer, 2024-02-14:
Updated this behaviour in this PR too: https://github.com/atopile/atopile/pull/143/

Now it will:
- Init and make an "init" commit in a clean repo when it's not already in a repo
- Completely remove the `.git` dir if it's in another repo