atopile/atopile #107: Could not find package 'regulators' in registry.

m5p3nc3r, 2024-02-07:
When issuing the command 'ato install regulators', I am getting an error response

```bash
ERROR    Error                                                                              errors.py:201
         --> Could not find package 'regulators' in registry.   
```
I am assuming this is an error as the regulators package is used in the projects README.md examples

mawildoer, 2024-02-07:
Sorry about this! We accidentally corrupted this entry in our db working to upgrade for the next package manager (coming soon - very excited!)

I've fixed it manually.

Thanks for the report! 🙏

closed 2024-02-07 by mawildoer

m5p3nc3r, 2024-02-07:
Hey @mawildoer, fully understand.  Just started playing with this today and I have to say I'm loving it so far!  Coder at heart, wanting to build some hardware.  This seems like a much more logical way for me to approach the solution :o)

Timot05, 2024-02-07:
Glad you like it @m5p3nc3r ! Keep us posted if you run into issues/have feedback.