atopile/atopile #114: Allow chaining of connections

m5p3nc3r, 2024-02-08:
This is a nice to have, but could help with boilerplate in definitions.

I want to create a chain of items (LED's) that all have power and ground requirements.  Currently I have to define it like this:
```
power = new Power
led1 = new LED
led2 = new LED

led1.power ~ power
led2.power ~ power
```

It would be nice to use syntax like this instead

```
led1.power ~ led2.power ~ power
```

mawildoer, 2024-02-08:
This'll be great when we get it in, and improve readability if we do it right!

Some of the reason we haven't yet is because we also want to support chaining serial connections and it needs to be super obvious whether you're doing a serial or parallel chain.

For series we've had two independent suggestions for: `a ~> b ~> c` (where `b` has an "in" and an "out" / is a dipole)
I quite like that, but it's very similar to the parallel chaining you've proposed.

For parallel what about something like `a ~ (b, c)` instead? That'd yield links `a ~ b` and `a ~ c`?

mawildoer, 2024-02-08:
Sorry - hit the wrong button! Definitely an open issue!

mawildoer, 2024-12-06:
@m5p3nc3r Possible in v0.3 with a fab module! I'll leave if open though, because I believe it should also be possible via ato

closed 2025-03-11 by napowderly