atopile/atopile #115: Arrays of objects

m5p3nc3r, 2024-02-08:
Thinking about reducing boilerplate as in #114, I would like to propose the inclusion of arrays and iterators to the language

In the below example, I create a chain of 16 ws2812b LED's and connect them up using iteration semantics:

```python
power = Power
leds = [WS2812B;16] # construct an array of 16 leds

# Connect all LEDS to the power
for led in leds:
    leds.power = power

# Chain the data lines
for (a, b) in zip(leds, leds[1:]:
    a.do | b.din
```

I believe this makes the code much more readable and reduces a lot of duplication where errors could arise.

I am not sure what relationship the language has with python (or any other base language), but it would be cool if we could use some of the collection semantics to help with code simplification.

mawildoer, 2024-02-08:
All great ideas! We haven't prioritised the for loops yet specifically because currently there's no procedural behaviour required for ato to build.

In the not-too-distant future, we'd definitely love something like this!

As long as we keep things relatively functional, so we can make good-quality assertions re. types etc... I'm certainly contented. The linking LEDs is one we've looked at before.

What do you think about: `leds[:-1].do ~ leds[1:].din` as well? I am unconvinced it's clearer, but to spitball.
It's also quite like numpy broadcasting which is something familiar to many (not sure how many EEs)

For the creation syntax, what about something like `leds = [WS2812] * 16` instead? Hopefully also very familiar to those with familiarity in Python!

Polsaker, 2024-02-08:
I'm not too familiar with numpy but I really dig that compact syntax, and the array/list multiplication also feels better than `[WS2812B;16]` imo

m5p3nc3r, 2024-02-08:
Sorry, I was mixing my syntax.  `[WS2812B; 16]` is a rust style initialiser.  I'm not a python expert, but I believe python would look more like `leds = [WS2812B() for i in range(16)]`.

I wanted to ask the question why are you creating a new language specification for this project.  I'm sure you have your reasons, but why are you not able to just use Python syntax and then you can inherit all the tools (language server, copilot, ...) that comes with it?

I do like the compact nature of `leds[:-1].do] ~ leds[1:].din`, but it has the disadvantage of being a new syntax that you would have to train the copilots and linters etc  of the world on before power tools in IDE's would be useable.

Maybe this deserves another issue for discussion, and I'm happy to start a new discussion, but would something pure python work for you here.  The interesting thing is that copilot did most of the heavy lifting in the following example below for me.  Unless you have a strategic need, or indeed an academic interest,  in creating a new language for schematic capture, is it worth a discussion on something like this:

```python
# Base class for all components
class Component():
    footprint: str
    designator_prefix: str
    mpn: str
    type: str

    def __init__(self):
        self.designator_prefix = "<undefined>"
        self.mpn = "<undefined>"
        self.type = "<undefined>"

class AddressableLED(Component):
    power: Power = Power()
    din: signal = signal()
    do: signal = signal()
    def __init__(self, name):
        print("AddressableLED")

class WS2812B(AddressableLED):
    cap = Capacitor()
    def __init__(self):
       self.designator_prefix = "U"
        self.mpn = "WS2812B"
        self.type = "LED"

class LEDStrip(Module):
    def __init__(self):
        self.components = [WS2812B() for i in range(16)]
        for (a, b) in zip(self.components, self.components[1:]):
            a.do.connect_pin(b.din)

# Base class for all modules
class Module():
    components = []
    def __init__(self):
        print("TODO: Init Module")

    def add_component(self, component):
        self.comonents.append(component)

    def export_kicad_netlist(self):
        print("TODO: Export kicad netlist")

class LEDStrip(Module):
    def __init__(self):
        self.components = [WS2812B() for i in range(16)]
        for (a, b) in zip(self.components, self.components[1:]):
            a.do.connect_pin(b.din)

strip = LEDStrip()
strip.export_netlist()
```

mawildoer, 2024-02-08:
Haha we should definitely write a blog post about that - you're not the first to ask!
I am leaning harder towards more "python like", but there's a few reasons we have a DSL.

From Narayan's answer on HN https://news.ycombinator.com/item?id=39267692 [(whole thread)](https://news.ycombinator.com/item?id=39263854)
> We had a look at a few options before doing our own. A few thoughts; Building a DSL allows us to keep it really simple to start, > we tried using SKIDL ( a very cool python project) and found it was pretty easy to get your self in a weird spot, the language > would let users do things like create a component and not hold a reference to it.
> Units and tolerances are core to our language, the physical world is 'fuzzy' and having a good way to deal with those we think > is pretty important.
> 
> We are also trying to make it as readable and friendly as possible, our expectation is our users will likely have some > experience with python and perhaps a little C back in school, so making it clear and approachable is front of mind.
> 
> Very open to critiques on our choices! We still very much in development.

SKiDL is well worth checking out as well: https://github.com/devbisme/skidl. It's a great project and Dave, it's creator, is an awesome guy too.

Within the compiler itself, I'm wanting to create object models along the lines you're describing, because I think it'll allow for much richer drafting of plugins and "top-level-types"

Re. the syntax of, depending on your container classes behaviour (eg. in the case of `numpy` this works) `a[N] <operator> b[N]` often/usually works, but we have an alternative operator (`~`) and assumes that the containers have the same "shape".

I reckon it might also be more declarative and easier to read?

m5p3nc3r, 2024-02-09:
Awesome, I assumed there was a valid reason for your design decisions.  Sorry for the spam!

The initial ask in this issue is still valid though, arrays/lists and iterators in the language would be awesome :o)

mawildoer, 2024-02-09:
No spam at all! Love the discussion

mawildoer, 2024-12-06:
@m5p3nc3r this is possible in v0.3 with a fab module!

closed 2024-12-06 by mawildoer