atopile/atopile #123: No warning on double-importing the same component/module/interface

hollinsky, 2024-02-09:
There is certainly an argument that this is a "feature, not a bug" but if I
```
import Power from "generics/interfaces.ato"
import Power from "generics/interfaces.ato"
```
there is no warning.

mawildoer, 2024-02-09:
I'm faiiirly contented with this behaviour. I'd probably say we should add it to the linter once we have one worth a dime, but since it's not a big deal I'll de-prio it for the sec.

Thanks for the issue!

napowderly, 2025-03-11:
[image]

[image]

closed 2025-03-11 by napowderly