atopile/atopile #136: Internal `TypeError` on build

aleokdev, 2024-02-11:
Build fails with a TypeError on address.py:31. Using ato 0.2.10.

My main file:
```
import N096_2864KSWEG01_H30 from "N096_2864KSWEG01_H30.ato"
import Resistor from "resistors.ato"
import Capacitor from "capacitors.ato"

module MorseTx:
    signal gnd
    screen = new N096_2864KSWEG01_H30
    c1 = new Capacitor
    c1.value = 1uF +/- 10%
    c1.p1 ~ screen.C1P
    c1.p2 ~ screen.C1N
```
Everything else is default. Components imported: C2890596, C13622 (unused)

Build output:
```
(venv) X:\programming\se_project\hw\morse-tx>ato build -b default
[13:58:21] INFO     Using project X:\programming\se_project\hw\morse-tx                                                                                                                                                                                       INFO     Building default                                                                                                                                                                                                                          INFO     Writing outputs to X:\programming\se_project\hw\morse-tx\build                                                                                                                                                                            INFO     Building copy-footprints                                                                                                                                                                                                                  INFO     Building netlist                                                                                                                                                                                                               Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "X:\programming\se_project\hw\venv\Scripts\ato.exe\__main__.py", line 7, in <module>
  File "X:\programming\se_project\hw\venv\Lib\site-packages\click\core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "X:\programming\se_project\hw\venv\Lib\site-packages\click\core.py", line 1078, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "X:\programming\se_project\hw\venv\Lib\site-packages\click\core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "X:\programming\se_project\hw\venv\Lib\site-packages\click\core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "X:\programming\se_project\hw\venv\Lib\site-packages\click\core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "X:\programming\se_project\hw\venv\Lib\site-packages\atopile\errors.py", line 268, in wrapper
    return func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^
  File "X:\programming\se_project\hw\venv\Lib\site-packages\atopile\cli\common.py", line 134, in wrapper
    return f(*args, **kwargs, build_ctxs=build_ctxs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "X:\programming\se_project\hw\venv\Lib\site-packages\atopile\errors.py", line 268, in wrapper
    return func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^
  File "X:\programming\se_project\hw\venv\Lib\site-packages\atopile\cli\build.py", line 38, in build
    _do_build(build_ctx)
  File "X:\programming\se_project\hw\venv\Lib\site-packages\atopile\cli\build.py", line 76, in _do_build
    muster.targets[target_name](build_ctx)
  File "X:\programming\se_project\hw\venv\Lib\site-packages\atopile\cli\build.py", line 105, in wrapper
    return func(build_args)
           ^^^^^^^^^^^^^^^^
  File "X:\programming\se_project\hw\venv\Lib\site-packages\atopile\cli\build.py", line 133, in generate_netlist
    f.write(get_netlist_as_str(build_args.entry))
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "X:\programming\se_project\hw\venv\Lib\site-packages\atopile\netlist.py", line 156, in get_netlist_as_str
    netlist = builder.build(root)
              ^^^^^^^^^^^^^^^^^^^
  File "X:\programming\se_project\hw\venv\Lib\site-packages\atopile\netlist.py", line 121, in build
    all_components = list(filter(match_components, all_descendants(root)))
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "X:\programming\se_project\hw\venv\Lib\site-packages\atopile\instance_methods.py", line 44, in all_descendants
    for child in get_children(addr):
  File "X:\programming\se_project\hw\venv\Lib\site-packages\atopile\instance_methods.py", line 10, in get_children
    root_instance = lofty.get_instance_tree(root_addr)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "X:\programming\se_project\hw\venv\Lib\site-packages\atopile\front_end.py", line 895, in get_instance_tree
    if address.get_instance_section(addr):
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "X:\programming\se_project\hw\venv\Lib\site-packages\atopile\address.py", line 31, in wrapper
    return ":".join([drive_letter, func(remainder, *args, **kwargs)])
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: sequence item 1: expected str instance, NoneType found
```

mawildoer, 2024-02-13:
Hey mate! Firstly - apologies in the delay getting back to you. There shouldn't ever expect errors like this from bugs in `ato` code, this is something with the compiler - thank you for bringing it to our attention!

None of us on the core team currently use Windows machines. We're getting one setup to smoke/debug things when issues are reported, but if in the meantime, this PR has what I believe the fix is: https://github.com/atopile/atopile/pull/137

If you are using a [development install](https://atopile.io/getting-started/#atopile-compiler-with-git) you can just check out that branch and things should work.

mawildoer, 2024-02-13:
Okay - scratch that, I just merged it and it's released in `v0.2.11`. Please let me know if the issues persists and I'll reopen the issue.

Happy coding!

closed 2024-02-13 by mawildoer