atopile/atopile #139: Publish extension on Open VSX

Polsaker, 2024-02-13:
Some linux distros default to using this store instead of Microsoft's. See https://github.com/microsoft/vscode/issues/107997#issuecomment-703245178

mawildoer, 2024-02-13:
Thanks for bringing this up!
How painful is it to install from the Microsoft store for people on these distros? It kinda seems odd that it wouldn't provide compatibility for the regular, defacto store

Polsaker, 2024-02-13:
According to the [Arch wiki](https://wiki.archlinux.org/title/Visual_Studio_Code#Extensions_support), the easiest way is to download the file from the Microsoft store and install it manually. You won't get updates this way (gotta re-install using the same method each time a new version is published).

I just tried this and had to go to https://marketplace.visualstudio.com/items?itemName=atopile.atopile, version history tab and download the latest file from there

mawildoer, 2024-02-13:
I see. Alright, I'll see what's involved in getting it posted there

mawildoer, 2024-02-13:
Done: https://open-vsx.org/extension/atopile/atopile

closed 2024-02-13 by mawildoer