atopile/atopile #150: Internal TypeError on build

cos-overclock, 2024-02-15:
Build fails with a KeyError on instance_method.py:16. Using ato 0.2.12.
OS: Windows 11

My main file:

```
import Resistor from "generics/resistors.ato"

module SonicVoice:
    signal top
    signal out
    signal bottom

    r_top = new Resistor
    r_top.footprint = "R0603"
    r_top.value = 100kohm +/- 10%

    r_bottom = new Resistor
    r_bottom.footprint = "R0603"
    r_top.value = 200kohm +/- 10%

    top ~ r_top.p1; r_top.p2 ~ out
    out ~ r_bottom.p1; r_bottom.p2 ~ bottom
```

Build output:

```
[09:52:09] INFO     Using project D:\atopile\sonic-voice                                                                   common.py:73
           INFO     Building default                                                                    build.py:36
           INFO     Writing outputs to D:\atopile\sonic-voice\build                                                                    build.py:63
           INFO     Building copy-footprints                                                                    build.py:74
           INFO     Building netlist                                                                    build.py:74
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "D:\atopile\venv\Scripts\ato.exe\__main__.py", line 7, in <module>
  File "D:\atopile\venv\Lib\site-packages\click\core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\atopile\venv\Lib\site-packages\click\core.py", line 1078, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "D:\atopile\venv\Lib\site-packages\click\core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\atopile\venv\Lib\site-packages\click\core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\atopile\venv\Lib\site-packages\click\core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\atopile\venv\Lib\site-packages\atopile\errors.py", line 268, in wrapper
    return func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^
  File "D:\atopile\venv\Lib\site-packages\atopile\cli\common.py", line 134, in wrapper
    return f(*args, **kwargs, build_ctxs=build_ctxs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\atopile\venv\Lib\site-packages\atopile\errors.py", line 268, in wrapper
    return func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^
  File "D:\atopile\venv\Lib\site-packages\atopile\cli\build.py", line 38, in build
    _do_build(build_ctx)
  File "D:\atopile\venv\Lib\site-packages\atopile\cli\build.py", line 76, in _do_build
    muster.targets[target_name](build_ctx)
  File "D:\atopile\venv\Lib\site-packages\atopile\cli\build.py", line 105, in wrapper
    return func(build_args)
           ^^^^^^^^^^^^^^^^
  File "D:\atopile\venv\Lib\site-packages\atopile\cli\build.py", line 133, in generate_netlist
    netlist = builder.build(root)
              ^^^^^^^^^^^^^^^^^^^
  File "D:\atopile\venv\Lib\site-packages\atopile\netlist.py", line 121, in build
    all_components = list(filter(match_components, all_descendants(root)))
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\atopile\venv\Lib\site-packages\atopile\instance_methods.py", line 45, in all_descendants
    yield from all_descendants(child)
  File "D:\atopile\venv\Lib\site-packages\atopile\instance_methods.py", line 44, in all_descendants
    for child in get_children(addr):
  File "D:\atopile\venv\Lib\site-packages\atopile\instance_methods.py", line 16, in get_children
    nested_instance = nested_instance.children[child_ref]
                      ~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^
KeyError: 'D:top'
```

My atopile project directory is `D:\atopile`.

mawildoer, 2024-02-15:
Sorry for the delay!

I didn't immediately recognise the issue since I was getting hooked on the `D:` drive path rather than anything important.

I believe this fixes it: https://github.com/atopile/atopile/pull/153

cos-overclock, 2024-02-16:
Confirmed fixed.
Thank you.

closed 2024-02-16 by cos-overclock