atopile/atopile #155: `BlockNotFoundError` on build

cos-overclock, 2024-02-16:
Build fails with a lockNotFoundError. Using ato 0.2.14.
OS: Windows 11

My main file:
```
import Resistor from "generics/resistors.ato"

module SonicVoice:
    signal top
    signal out
    signal bottom

    r_top = new Resistor
    r_top.footprint = "R0603"
    r_bottom = new Resistor
    r_bottom.footprint = "R0603"

    top ~ r_top.p1; r_top.p2 ~ out
    out ~ r_bottom.p1; r_bottom.p2 ~ bottom
```

```
[09:11:44] INFO     Using project D:\atopile\sonic-voice                                                                                                                                                          common.py:73
           INFO     Building default                                                                                                                                                                               build.py:36
           INFO     Writing outputs to D:\atopile\sonic-voice\build                                                                                                                                                build.py:63
           INFO     Building copy-footprints                                                                                                                                                                       build.py:74
           INFO     Building netlist                                                                                                                                                                               build.py:74
           ERROR    BlockNotFoundError                                                                                                                                                                            errors.py:77
                    --> No block named D:\atopile\sonic-voice\elec\src\sonic-voice.ato:SonicVoice in \atopile\sonic-voice\elec\src\sonic-voice.ato

Unfortunately errors ^^^ stopped the build. If you need a hand jump on Discord! https://discord.gg/mjtxARsr9V 👋
```

I believe this is due to the contents of the `build/default.net` file being empty.
It built successfully in github actions.

mawildoer, 2024-02-16:
Firstly, sorry about the Windows bugs and thank you for baring with us!

Someone else reported this on our Discord: https://discord.com/channels/1022538123915300865/1207774309016412210/1207780757288783922

I'm seeing what I can figure out now

mawildoer, 2024-02-16:
I've got a patch up here: https://github.com/atopile/atopile/pull/156
I'm waiting on validation to merge it, although you can already checkout this branch if you want to keep going!

Thanks for your patience 🙏

mawildoer, 2024-02-16:
Been confirmed fixed! Thanks for the patience!

closed 2024-02-16 by mawildoer