atopile/atopile #170: Incorrect pin name for component pin obtained with `ato install`.

cos-overclock, 2024-02-21:
For example, if a component is acquired with the following command,

```
ato install --jlcpcb C478472
```

The pin names for this component are shown in the following image.

[image]

However, multiple signals have been merged as the same name, perhaps due to the substitution of special characters.

```
component TAS5805MPWPR:
    # component TAS5805MPWPR
    footprint = "HTSSOP-28_L9.7-W4.4-P0.65-LS6.4-BL-EP"
    lcsc_id = "C478472"
    mpn = "C478472"
    # pins
    signal DGND ~ pin 1
    signal DVDD ~ pin 2
    signal ADR_FAULT_ ~ pin 3
    signal VR_DIG ~ pin 4
    DGND ~ pin 5
    signal LRCLK ~ pin 6
    signal SCLK ~ pin 7
    signal SDIN ~ pin 8
    signal SDOUT ~ pin 9
    signal SDA ~ pin 10
    signal SCL ~ pin 11
    signal PDN_ ~ pin 12
    signal AVDD ~ pin 13
    signal AGND ~ pin 14
    signal PVDD ~ pin 15
    PVDD ~ pin 16
    signal OUT_B_ ~ pin 17
    signal BST_B_ ~ pin 18
    signal PGND ~ pin 19
    OUT_B_ ~ pin 20
    BST_B_ ~ pin 21
    signal BST_A_ ~ pin 22
    signal OUT_A_ ~ pin 23
    PGND ~ pin 24
    BST_A_ ~ pin 25
    OUT_A_ ~ pin 26
    PVDD ~ pin 27
    PVDD ~ pin 28
    signal EP ~ pin 29
```

Specifically, `OUT_A+` and `OUT_A-` are identical signals called `OUT_A_`.
The same is true for `BST_`.

Timot05, 2024-03-01:
Thanks for flagging! We'll have a look

Timot05, 2024-03-01:
This was just fixed in the project's fork. We'll publish it to pypi asap.

napowderly, 2024-03-15:
Sorry for the delay! Should be fixed in the latest release https://pypi.org/project/easyeda2ato/

closed 2024-03-15 by napowderly