atopile/atopile #180: KiCad plugin does not get installed to the correct location on Linux

hollinsky, 2024-02-24:
On Ubuntu, this location is set to `~/.local/share/kicad/8.0/scripting/plugins`. Probably there'll need to be some logic to find the latest KiCad to install the plugin into as well.

https://github.com/atopile/atopile/blob/7abd489a2582768b15de88055934fe8dfbb5d8d2/src/atopile/cli/configure.py#L112

redwoz, 2024-08-05:
...and when installing KiCad as a flatpak on Ubuntu 22.04, the location becomes `~/.var/app/org.kicad.KiCad/data/kicad/8.0/scripting/plugins/`.

Can anyone confirm exactly what the `plugin_path = ` should be set to in the `atopile.py` file which gets dropped into this location? I'm not having any luck manually adding this file to this location in Ubuntu 22.04... refreshing plugins doesn't result in the desired outcome of the atopile icons appearing in the tool ribbon in PCB Editor... =\

mawildoer, 2024-08-05:
According to this section in the KiCAD docs there are three candidate locations: https://dev-docs.kicad.org/en/apis-and-binding/pcbnew/index.html#_typical_plugin_structure

If there a synthetic home path for flatpak's or something? It will unfortunately be pretty hard to handle all these various possible install locations for various distros and distribution mechanisms

closed 2024-12-06 by mawildoer