atopile/atopile #186: Imports in imported file failes

sethfischer, 2024-02-28:
# Problem

Imports in imported file fails.

`generics/connectors.ato` fails to import `WJ15EDGRC_3_81_2P from "elec/src/WJ15EDGRC_3_81_2P.ato"`

# How to reproduce

Create ato project according to [Getting started](https://atopile.io/getting-started/).

```
$ ato --version
ato, version 0.2.21
```

## `ato.yml`

```
ato-version: ^0.2.0
builds:
  default:
    entry: elec/src/test-imports.ato:TestImports
dependencies:
- generics
```

## `elec/src/test-imports.ato:TestImports`

```
from "generics/connectors.ato" import Header2Pin

module TestImports:
    signal gnd

    header = new Header2Pin
```

## Build

Running `ato build` results in:

```
$ ato build
[20:13:32] INFO     Using project …/tmp/test-atopile-import/test-imports                              common.py:73
           INFO     Building default                                                                  build.py:36
           INFO     Writing outputs to …/tmp/test-atopile-import/test-imports/build                   build.py:64
           INFO     Building copy-footprints                                                          build.py:77
           INFO     Building netlist                                                                  build.py:77
           ERROR    ImportNotFoundError                                                               errors.py:78
                    …/tmp/test-atopile-import/test-imports/.ato/modules/generics/connectors.ato:1                 
                    --> File 'elec/src/WJ15EDGRC_3_81_2P.ato' not found.  
```

napowderly, 2024-02-28:
Thanks, ill take a look

napowderly, 2024-02-28:
Looks like actually a mistake in the import statement:
import WJ15EDGRC_3_81_2P from "elec/src/WJ15EDGRC_3_81_2P.ato"
import WJ15EDGRC_3_81_2P from "elec/src/WJ15EDGRC-3.81-2P.ato"

not actually used anyway, I have just deleted it and updated generics, tagging a new release for it now.

closed 2024-02-28 by napowderly