atopile/atopile #217: FileNotFoundError while copying footprints during build

napowderly, 2024-03-08:
ild.py:82
Traceback (most recent call last):
  File "/Users/narayanpowderly/Documents/atopile-workspace/.venv/bin/ato", line 8, in <module>
    sys.exit(cli())
             ^^^^^
  File "/Users/narayanpowderly/Documents/atopile-workspace/.venv/lib/python3.11/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/narayanpowderly/Documents/atopile-workspace/.venv/lib/python3.11/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "/Users/narayanpowderly/Documents/atopile-workspace/.venv/lib/python3.11/site-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/narayanpowderly/Documents/atopile-workspace/.venv/lib/python3.11/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/narayanpowderly/Documents/atopile-workspace/.venv/lib/python3.11/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/narayanpowderly/Documents/atopile-workspace/atopile/src/atopile/errors.py", line 270, in wrapper
    return func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/Users/narayanpowderly/Documents/atopile-workspace/atopile/src/atopile/cli/common.py", line 142, in wrapper
    return f(*args, **kwargs, build_ctxs=build_ctxs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/narayanpowderly/Documents/atopile-workspace/atopile/src/atopile/cli/build.py", line 38, in build
    _do_build(build_ctx)
  File "/Users/narayanpowderly/Documents/atopile-workspace/atopile/src/atopile/cli/build.py", line 84, in _do_build
    muster.targets[target_name](build_ctx)
  File "/Users/narayanpowderly/Documents/atopile-workspace/atopile/src/atopile/cli/build.py", line 113, in wrapper
    return func(build_args)
           ^^^^^^^^^^^^^^^^
  File "/Users/narayanpowderly/Documents/atopile-workspace/atopile/src/atopile/cli/build.py", line 132, in consolidate_footprints
    shutil.copy(fp, fp_target)
  File "/Users/narayanpowderly/.platformio/python3/lib/python3.11/shutil.py", line 419, in copy
    copyfile(src, dst, follow_symlinks=follow_symlinks)
  File "/Users/narayanpowderly/.platformio/python3/lib/python3.11/shutil.py", line 256, in copyfile
    with open(src, 'rb') as fsrc:
         ^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: '/Users/narayanpowderly/Documents/atopile-workspace/swoop/.ato/modules/sk6805-ec20/.ato/modules/generics/elec/footprints/C0805.kicad_mod'

mawildoer, 2024-03-11:
Have you seen this come up again?

mawildoer, 2024-04-25:
Closing due to inactivity

closed 2024-04-25 by mawildoer