atopile/atopile #245: How make a pub for a keyboard?

mamcx, 2024-03-12:
I wish to create a custom keyboard like https://github.com/LouWii/ErgoMax, It will be nice if exists a sample or tutorial to do this kind of project!

napowderly, 2024-03-12:
Looks cool! Have you checked out some of our getting started videos? Should get you setup with most of the basics, getting parts, making modules etc. Also, feel free to jump on our discord, I am sure you will find people interested in your project!

https://www.youtube.com/watch?v=eMWRwZOajdQ
https://discord.gg/6BEN8bY53G

mawildoer, 2024-03-14:
@mamcx are these the kind of tutorials you're looking for?

mamcx, 2024-03-14:
Yeah the tutorial is good. I think I need to consider how adapt it to a keyboard...

mawildoer, 2024-03-15:
Jump on discord if you’re looking for a hand! https://discord.gg/nkpC7wYRVw

closed 2024-03-15 by mawildoer