atopile/atopile #262: ato install {package} does not add dependency to ato.yaml

napowderly, 2024-03-20:
When installing to an older project without dependencies, or empty dependencies, causes error and does not add to deps:
[image]
[image]

mawildoer, 2024-04-09:
It kind of does, just in this case it's actually missing the `dependencies` key. There's probably a hacky fix for this, but the easiest thing is to include that key just empty as `dependencies: []`

napowderly, 2024-04-09:
Perhaps a better error message?

mawildoer, 2024-05-15:
Closing as duplicate

closed 2024-05-15 by mawildoer