atopile/atopile #278: assertions dont work with negative numbers

napowderly, 2024-04-09:
Want to be able to capture negative voltage/current rating

[image]
[image]

mawildoer, 2024-04-10:
I don't think this is an issue with the asserions, I think it's an issue with the ranged values parsing?
In that it looks like it has parsed the `-400mV to 400mV` as `400mV to 400mV`

mawildoer, 2024-04-19:
This is the section of the front-end responsible for handling these inversions; https://github.com/atopile/atopile/blob/eb7433b79ee0304cf0333b50719b071680109262/src/atopile/front_end.py#L640-L646

PR #314 (2024-04-25): Fix negative unaries
Closes #278 

Before, something like `-1 to 1` didn't work. It'd register as (1 to 1)
This fixes it and sliiiiiggghhhtly cleans up some of the junk parsing these ranged expressions
closed 2024-04-25 by mawildoer