atopile/atopile #365: Components are missing in the block diagram view

mawildoer, 2024-05-14:
Components aren't displayed in the block diagram view. I know it's meant to be a simplified view, but it's inaccurate to say there's nothing there. Instead, we should display them the same way as modules.

Then at least I'd know to switch to the schematic view

Example:
[image]
[image]

Timot05, 2024-05-20:
[image]
They are there. Just need to pan around. I'll see if I can fix.

Timot05, 2024-05-20:
@mawildoer mind checking if [this PR](https://github.com/atopile/atopile/pull/371) fixes it? It does for me.