atopile/atopile #403: Capacitor footprint 1206 does not resolve

ReinBentdal, 2024-06-28:
```
  cap = new Capacitor
  cap.footprint = "1206" # bug not finding cap? results in footprint 206 in debug
  cap.value = 100uF +/- 20%
  cap.p1 ~ vbat
  cap.p2 ~ gnd
```
results in
>ERROR    No component matches parameters                                                                                                                                                  errors.py:81
                    Address: bms.ato:Bms::battery_charger.connector.cap                                                                                                                                          
                        No valid component found for spec designator_prefix == C && mpn == generic_capacitor && type == capacitor && value == {'unit': 'microfarad', 'min_val': 80.00000000000001,               
                    'max_val': 120.00000000000003, 'nominal': 100.00000000000003, 'tolerance': 20.000000000000004, 'tolerance_pct': 20.0} && package == 206: please check the part specs above, if               
                    they look right, we probably dont have it yet, we are working on it!

As the error message highlights, the package is interpreted as _package == 206_.
If the footprint select line is removed, the cap actually resolves to a 1206 package.

napowderly, 2024-07-02:
We need to improve the errors significantly for the server, on our list. The footprint attribute directly matches the name of the footprint, in KiCADs case caps are C{size}. If you just want to specify the size, you can use the package attribute. 

This should work for you:
cap.package = "1206"

closed 2024-07-02 by napowderly

ReinBentdal, 2024-07-03:
Hi,
Think there is a slight misunderstanding. Your suggestion is what I wrote, and it seems like atopile interprets "1206" as "206".