atopile/atopile #409: `ato create` error, unable to delete .git *.idx file, access denied.

Marcus10110, 2024-07-02:
First time user.

This is Mark from Saleae, feel free to email me for more details if needed. I'll give this a try on my Mac in the meantime.

Environment:
Windows 10 x64
Python 3.12.4

Steps to reproduce:
```
python -m venv .venv
.venv\Scripts\activate
pip install atopile
ato create
```
create project, do not create github repo, project name "test"

Output:
```
fatal: not a git repository (or any of the parent directories): .git
What do you want to create [project/build]: project
🚀 What's your project name?: test
Would you like to create a new repo for this project? [y/n]: n
[21:45:25] INFO     Installing  in C:\Users\markg\Software\Temp\atopile\test                                                                                                                                                install.py:67
[21:45:26] INFO     Installing dependency generics                                                                                                                                                                         install.py:167
[21:45:27] INFO     Using ✨ v3.0.5 ✨ of generics                                                                                                                                                                         install.py:232
           INFO     Done! 🤙                                                                                                                                                                                               install.py:120
           INFO     Installing generics in C:\Users\markg\Software\Temp\atopile\test                                                                                                                                        install.py:67
           INFO     Installing dependency generics                                                                                                                                                                         install.py:167
[21:45:28] INFO     Using ✨ v3.0.5 ✨ of generics                                                                                                                                                                         install.py:232
Traceback (most recent call last):
  File "C:\Users\markg\AppData\Local\Programs\Python\Python312\Lib\shutil.py", line 633, in _rmtree_unsafe
    os.unlink(fullname)
PermissionError: [WinError 5] Access is denied: 'C:\\Users\\markg\\Software\\Temp\\atopile\\test\\elec\\src\\generics\\.git\\objects\\pack\\pack-b855b32e1f2590757cf75549474f5ec8d38fb96d.idx'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\markg\Software\Temp\atopile\.venv\Scripts\ato.exe\__main__.py", line 7, in <module>
  File "C:\Users\markg\Software\Temp\atopile\.venv\Lib\site-packages\click\core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\markg\Software\Temp\atopile\.venv\Lib\site-packages\click\core.py", line 1078, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "C:\Users\markg\Software\Temp\atopile\.venv\Lib\site-packages\click\core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\markg\Software\Temp\atopile\.venv\Lib\site-packages\click\core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\markg\Software\Temp\atopile\.venv\Lib\site-packages\click\core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\markg\Software\Temp\atopile\.venv\Lib\site-packages\atopile\cli\create.py", line 216, in create
    do_install(
  File "C:\Users\markg\Software\Temp\atopile\.venv\Lib\site-packages\atopile\cli\install.py", line 98, in do_install
    robustly_rm_dir(abs_path / ".git")
  File "C:\Users\markg\Software\Temp\atopile\.venv\Lib\site-packages\atopile\utils.py", line 14, in robustly_rm_dir
    shutil.rmtree(path, onerror=remove_readonly)
  File "C:\Users\markg\AppData\Local\Programs\Python\Python312\Lib\shutil.py", line 781, in rmtree
    return _rmtree_unsafe(path, onexc)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\markg\AppData\Local\Programs\Python\Python312\Lib\shutil.py", line 635, in _rmtree_unsafe
    onexc(os.unlink, fullname, err)
  File "C:\Users\markg\AppData\Local\Programs\Python\Python312\Lib\shutil.py", line 750, in onexc
    return onerror(func, path, exc_info)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\markg\Software\Temp\atopile\.venv\Lib\site-packages\atopile\utils.py", line 12, in remove_readonly
    func(path)
PermissionError: [WinError 5] Access is denied: 'C:\\Users\\markg\\Software\\Temp\\atopile\\test\\elec\\src\\generics\\.git\\objects\\pack\\pack-b855b32e1f2590757cf75549474f5ec8d38fb96d.idx'
           INFO     Ignored error after process had died: OSError(9, 'The handle is invalid', None, 6, None)                                                                                                                   cmd.py:773
           INFO     Ignored error after process had died: OSError(9, 'The handle is invalid', None, 6, None)
```

Taking a quick look, the file in question does exist. Also, the permissions appear normal, and I can delete it as a regular user. Here are all the files in that directory (below done with bash, above commands tested on windows CMD and windows PowerShell)
```
find elec/src/generics/.git/
elec/src/generics/.git/
elec/src/generics/.git/config
elec/src/generics/.git/description
elec/src/generics/.git/HEAD
elec/src/generics/.git/hooks
elec/src/generics/.git/index
elec/src/generics/.git/info
elec/src/generics/.git/logs
elec/src/generics/.git/objects
elec/src/generics/.git/objects/info
elec/src/generics/.git/objects/pack
elec/src/generics/.git/objects/pack/pack-b855b32e1f2590757cf75549474f5ec8d38fb96d.idx
elec/src/generics/.git/objects/pack/pack-b855b32e1f2590757cf75549474f5ec8d38fb96d.pack
elec/src/generics/.git/packed-refs
elec/src/generics/.git/refs
elec/src/generics/.git/refs/heads
elec/src/generics/.git/refs/heads/main
elec/src/generics/.git/refs/remotes
elec/src/generics/.git/refs/remotes/origin
elec/src/generics/.git/refs/remotes/origin/HEAD
elec/src/generics/.git/refs/tags
```

mawildoer, 2024-07-02:
Hey @Marcus10110 ! Thanks for the issue and appreciate the details!

I'm not 100% sure what's going wrong, but I've caught the exception and downgraded it to a warning in the log: https://github.com/atopile/atopile/pull/410

Marcus10110, 2024-07-02:
Awesome, thanks for the super fast turnaround! I built and installed main branch locally and have confirmed the error is now reported as a warning, and `ato build` works on the empty default project!

```
(.venv) C:\Users\markg\Software\Temp\atopile_test>ato create
What do you want to create [project/build]: project
🚀 What's your project name?: test
Would you like to create a new repo for this project? [y/n]: n
[23:13:04] INFO     Installing  in C:\Users\markg\Software\Temp\atopile_test\test                                                                                                                               install.py:67
[23:13:06] INFO     Installing dependency generics                                                                                                                                                             install.py:172
[23:13:07] INFO     Using ✨ v3.0.5 ✨ of generics                                                                                                                                                             install.py:237
           INFO     Done! 🤙                                                                                                                                                                                   install.py:125
           INFO     Installing generics in C:\Users\markg\Software\Temp\atopile_test\test                                                                                                                       install.py:67
           INFO     Installing dependency generics                                                                                                                                                             install.py:172
[23:13:08] INFO     Using ✨ v3.0.5 ✨ of generics                                                                                                                                                             install.py:237
           WARNING  Error                                                                                                                                                                                        errors.py:81
                        Failed to remove .git directory: PermissionError(13, 'Access is denied')
           INFO     Done! 🤙                                                                                                                                                                                   install.py:125
✨ Created new project "test"! ✨
           INFO     Ignored error after process had died: OSError(9, 'The handle is invalid', None, 6, None)                                                                                                       cmd.py:773
           INFO     Ignored error after process had died: OSError(9, 'The handle is invalid', None, 6, None)                                                                                                       cmd.py:786
```

PR #410 (2024-07-02): Fix `ato create`
Fixes #409
closed 2024-07-02 by Marcus10110