atopile/atopile #430: Docs should elevate information on adding footprints

jbullion-astra, 2024-07-17:
Currently the info is under the [install dependencies](https://atopile.io/install/#adding-custom-footprints). This information should either be promoted to its own page, or at least have a link in the how to section.

mawildoer, 2024-07-18:
Thanks for the suggestion! You're right - this is all a little obfuscated. Does this explain what you're looking for? https://github.com/atopile/atopile/pull/431

jjrreett, 2024-07-18:
It’s certainly better than it was! Might want to link or update the install page. 

I have been dropping my 'kicad_mod' files into the build directory. My take on your docs is that any mod file in src will get picked up?

This would also be a good place to note how pin naming correlates with the footprints.

mawildoer, 2024-07-18:
Yeah, great point. Made those updates as well. Thanks for raising it! 🙌

PR #431 (2024-07-18): Add how-to on installing footprints
Fixes #430
commit 5a4e2323 (2024-07-18): Add how-to on installing footprints

Fixes #430