atopile/atopile #433: Enforce lower-level declaration for overrides

mawildoer, 2024-07-18:
You shouldn't be able to assign arbitrary values to a class that isn't "self". That is, `a.b.c = 1` should fail if `c` isn't already declared for `a.b`.

sethfischer, 2024-07-20:
After upgrading from v0.2.57 to v0.2.58 the following ato code:

```
from "ams1117-33/elec/src/ams1117-33.ato" import AMS111733  # ato install ams1117-33

module MyModule:
    ⋮
    # 3V3 regulator
    ldo_3v3 = new AMS111733
    ldo_3v3.power_in ~ power_connector.power
    ldo_3v3.power_out ~ power_3v3
```

Generates the following warnings:

```
[14:33:52] WARNING  Implicit Declaration Future Deprecation Warning                     errors.py:80
    .../.ato/modules/ams1117-33/elec/src/ams1117-33.ato:32:4
    Field 'value' not declared for .../elec/src/entry.ato:MyModule::ldo_3v3.input_cap.
    Declaring implicitly for now, but in the future this may become an error...
WARNING  Implicit Declaration Future Deprecation Warning                                errors.py:80
    .../.ato/modules/ams1117-33/elec/src/ams1117-33.ato:34:4
    Field 'package' not declared for .../elec/src/entry.ato:MyModule::ldo_3v3.input_cap.
    Declaring implicitly for now, but in the future this may become an error...
WARNING  Implicit Declaration Future Deprecation Warning                                errors.py:80
    .../.ato/modules/ams1117-33/elec/src/ams1117-33.ato:37:4
    Field 'value' not declared for .../elec/src/entry.ato:MyModule::ldo_3v3.output_cap.
    Declaring implicitly for now, but in the future this may become an error...
WARNING  ... forgoing more "Implicit Declaration Future Deprecation Warning" errors of  errors.py:193
```

sethfischer, 2024-07-20:
```
$ ato --version
ato, version 0.2.58
```

The following specifies the main power input voltage and also the allowed input voltage of a sub module. This ato code did not generate any warnings or errors in version 0.2.57.

```
from "generics/interfaces.ato" import Power

module SubModule:
    power = new Power
    # specify allowable voltage for this sub module
    power.voltage = 3.1V to 3.3V

module Main:
    power = new Power
    # specify main power input voltage
    power.voltage = 3.1V to 3.3V

    sub_module = new SubModule
    sub_module.power ~ power
```

Version 0.2.58 generates the following warnings and error:

```
$ ato build
[15:35:40] INFO     Using project .../atopile-433/test              common.py:71
    INFO     Building default                                       build.py:38
    WARNING  Implicit Declaration Future Deprecation Warning        errors.py:80
    .../atopile-433/test/elec/src/test.ato:13:4
    Field 'voltage' not declared for .../atopile-433/test/elec/src/test.ato:Main::power.
    Declaring implicitly for now, but in the future this may become an error...
    WARNING  Implicit Declaration Future Deprecation Warning        errors.py:80
    .../atopile-433/test/elec/src/test.ato:7:4
    Field 'voltage' not declared for .../atopile-433/test/elec/src/test.ato:Main::sub_module.power.
    Declaring implicitly for now, but in the future this may become an  error...
    ERROR Error                                                     errors.py:80
    .../atopile-433/test/elec/src/test.ato:16:4
    The source and target separately defined values for the attribute "voltage"
```

sethfischer, 2024-07-20:
## Setting `do_not_populate` on a component in a third-party package generates a deprecation warning

It may be necessary to set the `do_not_populate` (or `exclude_from_bom`) on a component provided by a third-party package. These two attributes were introduced in #419.

Version 0.2.58 indicates that this feature will be deprecated. It is likely not desired nor reasonable to expect third-party packages to explicitly define `do_not_populate` and `exclude_from_bom`.

### Environment

```
$ ato --version
ato, version 0.2.58
```

### How to reproduce

```
from "xt-connectors/xt-connectors.ato" import XT30_Male_Right_Angle

module Main:
    connector = new XT30_Male_Right_Angle
    connector.do_not_populate = True
```

### Run `ato build`

When the project is built the following warning is generated:

```
$ ato build
[11:02:03] INFO     Using project .../atopile-433/test                                                                   common.py:71
           INFO     Building default                                                                                     build.py:38
           WARNING  Implicit Declaration Future Deprecation Warning                                                      errors.py:80
                    .../atopile-433/test/elec/src/test.ato:5:4       
                        Field 'do_not_populate' not declared for .../atopile-433/test/elec/src/test.ato:Main::connector.
                        Declaring implicitly for now, but in the future this may become an error...
```

mawildoer, 2024-07-21:
Thanks for the reports @sethfischer . Don't stress too much. There's a bunch of places we should upgrade to avoid this warning, and it'll gradually disappear. Nothing to be concerned of for the minute

closed 2025-03-11 by napowderly