atopile/atopile #457: Error when passing values around with replace operator

napowderly, 2024-08-05:
I want to be able to specify a load capacitance in a given crystal component definition like so:
[image]

Then use that value to pass along to the generic 'oscillator' class to set the cap values
[image]

Replacement is done one level up:
[image]

[image]

**Replicate by building swift wing on commit below:**
https://github.com/atopile/swift-wing
commit: bc61bfa4cd4057ff20e84fd057bdf2d97142ea01

napowderly, 2024-08-05:
Figured it out, I have bad spelling for oscillator. Could use a better error though!

closed 2024-08-05 by napowderly

mawildoer, 2024-08-05:
It should also throw another error for trying to dot-access an object that doesn't exist. No error thrown in the log for the `oscilator.crystal` line?