atopile/atopile #536: Build resets component positions to origin

napowderly, 2024-11-18:
### Current Behaviour

When I run the build, all the components are moved back to the origin.

def main():
    logger.info("Building app")
    app = App()

    logger.info("Export")
    apply_design_to_pcb(app)

if __name__ == "__main__":
    setup_basic_logging()
    logger.info("Running example")

    typer.run(main)

### Expected Behaviour

Component positions are preserved

### Possible Solution

#pr number

### Version

4.1.3

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

mawildoer, 2024-11-19:
Confirmed the same behaviour for the examples, unfortunately.

iopapamanoglou, 2024-11-19:
`apply_design_to_pcb` is exclusively for examples (doesn't preserve positions)
Just like the rest in `libs/examples/**.py`
See `minimal_led_orderable.py` or `project-template` for how to build non-examples.

TLDR: Use `apply_design` instead. That one moves only modules that have a manual `has_pcb_position` trait and no `FBRK:notouch` flag

closed 2024-11-19 by iopapamanoglou