atopile/atopile #542: Terminate() behavior unclear

napowderly, 2024-11-19:
Tried to use i2c.terminate(), resistors do not have values set and there is not an obvious way to do so.

iopapamanoglou, 2024-11-19:
Definitely no obvious way at the moment indeed. Do you have a suggestion for how that should look?
How do people normally decide on their pulls for something like this?

Here is how it's done atm:
Explicit:
```python
i2c.terminate()
for line in [i2c.sda, i2c.scl]:
  for r in line.get_trait(F.ElectricLogic.has_pulls).get_pulls():
    r.resistance.merge(...)
```
Implicit:
```python
i2c.terminate()
for pull in [r for line in i2c.get_children(direct=False; types=(F.ElectricLogic,) if line.has_trait(F.ElectricLogic.has_pulls) for r in line.get_trait(F.ElectricLogic.has_pulls).get_pulls() ]:
    pull.resistance.merge(...)
```

napowderly, 2024-11-19:
It's just an RC delay to determine the minimum value, so you need to calculate bus capacitance (each device will have a spec) and an estimate for trace capacitance. The i2c spec is for <400pF. The required time constant is dependent on the switching frequency.

The maximum value is based on the device's maximum sink current and a practical consideration to limit power consumption.

Good TI doc on it:
https://www.ti.com/lit/an/slva689/slva689.pdf?ts=1732023268243&ref_url=https%253A%252F%252Fwww.google.com%252F

napowderly, 2024-11-19:
The above snipped mostly worked, I was getting some 'none' returns in the second for loop, ended up getting it to work like this:

        self.I2C.terminate()
        for line in [self.I2C.sda, self.I2C.scl]:
            for r in line.get_trait(F.ElectricLogic.has_pulls).get_pulls():
                if r is None:
                    continue
                r.resistance.merge(F.Range.from_center_rel(4.7 * P.kohm, 0.03))
                r.add(F.has_footprint_requirement_defined([("0201", 2)]))

closed 2025-03-11 by napowderly