atopile/atopile #546: Meta: Reopen or close PRs from other repo

iopapamanoglou, 2024-11-19:
### Feature Request

- [ ] https://github.com/atopile/faebryk/pull/132 @napowderly 
- [ ] https://github.com/atopile/faebryk/pull/91 @ruben-iteng 
- [ ] https://github.com/atopile/faebryk/pull/78 @ruben-iteng 
- [ ] https://github.com/atopile/faebryk/pull/73 @mawildoer 
- [ ] https://github.com/atopile/faebryk/pull/39 @mawildoer 
- [ ] https://github.com/atopile/faebryk/pull/31 @jrhrsmit 
- [ ] https://github.com/atopile/faebryk/pull/28 @io

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

ruben-iteng, 2024-12-14:
Done: https://github.com/atopile/atopile/pull/691

@sam-mellor might want to take a look at https://github.com/atopile/faebryk/pull/78

ruben-iteng, 2025-02-02:
Fixes old NetTie pr (https://github.com/atopile/faebryk/pull/39)

https://github.com/atopile/atopile/pull/924

https://github.com/atopile/atopile/pull/925