atopile/atopile #547: Improve full-names of nodes

mawildoer, 2024-11-19:
- [ ] Remove `*.` prefix from the start of things
- [ ] Remove root node's name (It's the same for everything within a design)

iopapamanoglou, 2024-11-20:
> Remove root node's name (It's the same for everything within a design)
That's not true for non-bound objects

> Remove *. prefix from the start of things
That doesn't exist anymore right?
Thats now hex(id)[-4:] or something similar

iopapamanoglou, 2024-11-21:
I have an idea!

- Let's make a 'is_application_root' trait. 
- Add in the c++ Node a flag, for skip_from_full_name (workaround because C++ has no access to trait atm)
- make the trait set that flag on obj_set
- check in 'get_full_name' whether any node contains the skip and skip the respective nodes from the name

commit 894b56e5 (2024-11-27): Improve get_full_name to remove prefixed *HEX on nodes with `is_app_root` trait

Fixes: #547