atopile/atopile #595: Build fails if components do not have an MPN

sethfischer, 2024-11-29:
### Current Behaviour

```console
$ ato --version
ato, version 0.2.69
```

When running `ato build` and if a component does not have an MPN the following error is encountered:

```
File ".../lib/python3.11/site-packages/atopile/bom.py", line 225, in generate_bom
_add_row(
TypeError: generate_bom.<locals>._add_row() missing 1 required positional argument: 'price'
```

### Expected Behaviour

If a part does not have a MPN then its price should be a empty string in the BoM.

### Possible Solution

Allow a default value of `None` for `price`.

`generate_bom._add_row` expects `price` as a required parameter.

https://github.com/atopile/atopile/blob/ef902b7b931e52b838e4c2920477aed527d6af9f/src/atopile/bom.py#L178

In the case that a part does not have a MPN, the call to `generate_bom._add_row` does not provide the required parameter `price`.

https://github.com/atopile/atopile/blob/ef902b7b931e52b838e4c2920477aed527d6af9f/src/atopile/bom.py#L225-L230

Suggested method signature for `generate_bom._add_row`:

```python
def _add_row(value, designator, footprint, mpn, price = None):
```

From https://docs.python.org/3/library/csv.html#csv.writer

> To make it as easy as possible to interface with modules which implement the DB API, the value None is written as the empty string.

### Version

0.2.69

### Relevant log output

```shell
+ Exception Group Traceback (most recent call last):
  |   File "…/bin/ato", line 8, in <module>
  |     sys.exit(cli())
  |              ^^^^^
  |   File "…/lib/python3.11/site-packages/click/core.py", line 1157, in __call__
  |     return self.main(*args, **kwargs)
  |            ^^^^^^^^^^^^^^^^^^^^^^^^^^
  |   File "…/lib/python3.11/site-packages/click/core.py", line 1078, in main
  |     rv = self.invoke(ctx)
  |          ^^^^^^^^^^^^^^^^
  |   File "…/lib/python3.11/site-packages/click/core.py", line 1688, in invoke
  |     return _process_result(sub_ctx.command.invoke(sub_ctx))
  |                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  |   File "…/lib/python3.11/site-packages/click/core.py", line 1434, in invoke
  |     return ctx.invoke(self.callback, **ctx.params)
  |            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  |   File "…/lib/python3.11/site-packages/click/core.py", line 783, in invoke
  |     return __callback(*args, **kwargs)
  |            ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  |   File "/usr/lib/python3.11/contextlib.py", line 80, in inner
  |     with self._recreate_cm():
  |   File "/usr/lib/python3.11/contextlib.py", line 158, in __exit__
  |     self.gen.throw(typ, value, traceback)
  |   File "…/lib/python3.11/site-packages/atopile/errors.py", line 351, in muffle_fatalities
  |     raise ex
  | ExceptionGroup:  (1 sub-exception)

  +-+---------------- 1 ----------------
    | Traceback (most recent call last):
    |   File "…/lib/python3.11/site-packages/atopile/errors.py", line 339, in muffle_fatalities
    |     yield
    |   File "/usr/lib/python3.11/contextlib.py", line 81, in inner
    |     return func(*args, **kwds)
    |            ^^^^^^^^^^^^^^^^^^^
    |   File "/usr/lib/python3.11/contextlib.py", line 81, in inner
    |     return func(*args, **kwds)
    |            ^^^^^^^^^^^^^^^^^^^
    |   File "…/lib/python3.11/site-packages/atopile/cli/common.py", line 139, in wrapper
    |     return f(*args, **kwargs, build_ctxs=build_ctxs)
    |            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    |   File "…/lib/python3.11/site-packages/atopile/cli/build.py", line 40, in build
    |     _do_build(build_ctx)
    |   File "…/lib/python3.11/site-packages/atopile/cli/build.py", line 102, in _do_build
    |     muster.targets[target_name](build_ctx)
    |   File "…/lib/python3.11/site-packages/atopile/cli/build.py", line 189, in generate_bom
    |     f.write(atopile.bom.generate_bom(build_args.entry))
    |             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    |   File "…/lib/python3.11/site-packages/atopile/bom.py", line 225, in generate_bom
    |     _add_row(
    | TypeError: generate_bom.<locals>._add_row() missing 1 required positional argument: 'price'
    +------------------------------------
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

sethfischer, 2024-11-30:
I can work on a MR for this if someone would like to assign it to me.

mawildoer, 2024-12-02:
Thanks for the issue. We've been heads down on v0.3!

If it's critical, happy to accept a MR now, but this code is all being replaced come v0.3, so perhaps it's worth implementing on the other side of that. Code's available now on `main`, but building all the way from `ato` stably w/ existing projects is under works

mawildoer, 2024-12-03:
Thank you @sethfischer ! PR merged

Wild that's all it took to fix it. Issue done?

sethfischer, 2024-12-03:
@mawildoer Yes done! thanks for merging

PR #609 (2024-12-03): Fix BOM price error when no MPN
# Fix BOM price error when no MPN

# Description

Allow creation of BOM when a part has no MPN and hence no price.

Fixes #595 

# Checklist

Please read and execute the following:

- [X] My code follows the [coding guidelines](/faebryk/faebryk/blob/main/docs/CODING_GUIDELINES.md) of this project
- [X] My PR title is following the [contribution guidelines](/faebryk/faebryk/blob/main/docs/CONTRIBUTING.md)
- [X] I have performed a self-review of my own code
- [ ] I have commented my code, particularly in hard-to-understand areas
- [X] I have made corresponding cha [... cut]
closed 2024-12-03 by sethfischer