atopile/atopile #712: 3D model path broken with ato build

iopapamanoglou, 2024-12-18:
### Current Behaviour

```
(model "${KIPRJMOD}/../libs/3dmodels/lcsc.3dshapes/R0603.wrl"
```

```
root
- standalone
  - default.kicad_pcb
- build
  - kicad
    - libs
      - 3dmodels
```

### Expected Behaviour

.

### Version

4d5ce269bc30147f102792c159515033160744f5

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

hexdae, 2025-01-03:
Hey @iopapamanoglou, maybe this is a different issue, but with `ato` v0.2.0 you could place a `.step` or `.wrl` model in the `footprints` directory and `ato build` would copy this under `build/footprints/footprints.3dshapes`

We have found that in ato 0.3.0 this is not the case anymore. Is this intended behavior or a bug?

hexdae, 2025-01-03:
I think this was the PR that introduce the behavior we relied upon https://github.com/atopile/atopile/pull/282

ruben-iteng, 2025-01-04:
Correct path: `${KIPRJMOD}/../../../build/kicad/libs/3dmodels/lcsc.3dshapes/<file_name>.wrl`
Also, idk if the footprints in `build/kicad/libs/footprints/lcsc.pretty` should contain a reference to the 3D files, they don't now

iopapamanoglou, 2025-01-06:
Reopening to check @hexdae 's issue.

Thanks @ruben-iteng for fixing this.

PR #789 (2025-01-05): Fix footprint 3D file path
# Fix footprint 3D file path

## Description

This works for now, but as stated in the [TODO](https://github.com/atopile/atopile/blob/a0a8acc6358c0ff7629fa581248b0fdbd4bd8f0e/src/atopile/cli/build.py#L91), should use buildcontext instead.

Fixes #712 

## Checklist

Please read and execute the following:

- [ ] My code follows the [coding guidelines](/faebryk/faebryk/blob/main/docs/CODING_GUIDELINES.md) of this project
- [ ] I have performed a self-review of my own code
- [ ] I have commented my code, particularly in hard-to-understand areas
- [ ] I have made corresp [... cut]