atopile/atopile #776: Kivas pcb file format filled zones island bugged

ruben-iteng, 2025-01-02:
### Current Behaviour

KiCad throw a file error when there is a zone without er but with the `island` flag. 

### Expected Behaviour

KiCad loads the file

### Possible Solution

Flag might be at the wrong position (after points, instead of above)

### Version

0.3.0-dev

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

mawildoer, 2025-01-03:
Thanks for the report @ruben-iteng ! Any chance you can attach a file exhibiting this issue?

ruben-iteng, 2025-01-04:
Yes of course :smile: 

(`filled_polygon` on the `F.Cu` layer is correct, the one on the `B.Cu` layer gives a file error)

[image]

<details>
  <summary>KiCad PCB file</summary>

```sexp
(kicad_pcb
	(version 20240108)
	(generator "pcbnew")
	(generator_version "8.0")
	(general
		(thickness 1.6)
		(legacy_teardrops no)
	)
	(paper "A4")
	(layers
		(0 "F.Cu" signal)
		(31 "B.Cu" signal)
		(32 "B.Adhes" user "B.Adhesive")
		(33 "F.Adhes" user "F.Adhesive")
		(34 "B.Paste" user)
		(35 "F.Paste" user)
		(36 "B.SilkS" user "B.Silkscreen")
		(37 "F.SilkS" user "F.Silkscreen")
		(38 "B.Mask" user)
		(39 "F.Mask" user)
		(40 "Dwgs.User" user "User.Drawings")
		(41 "Cmts.User" user "User.Comments")
		(42 "Eco1.User" user "User.Eco1")
		(43 "Eco2.User" user "User.Eco2")
		(44 "Edge.Cuts" user)
		(45 "Margin" user)
		(46 "B.CrtYd" user "B.Courtyard")
		(47 "F.CrtYd" user "F.Courtyard")
		(48 "B.Fab" user)
		(49 "F.Fab" user)
		(50 "User.1" user)
		(51 "User.2" user)
		(52 "User.3" user)
		(53 "User.4" user)
		(54 "User.5" user)
		(55 "User.6" user)
		(56 "User.7" user)
		(57 "User.8" user)
		(58 "User.9" user)
	)
	(setup
		(pad_to_mask_clearance 0)
		(allow_soldermask_bridges_in_footprints no)
		(pcbplotparams
			(layerselection 0x00010fc_ffffffff)
			(plot_on_all_layers_selection 0x0000000_00000000)
			(disableapertmacros no)
			(usegerberextensions no)
			(usegerberattributes yes)
			(usegerberadvancedattributes yes)
			(creategerberjobfile yes)
			(dashed_line_dash_ratio 12.000000)
			(dashed_line_gap_ratio 3.000000)
			(svgprecision 4)
			(plotframeref no)
			(viasonmask no)
			(mode 1)
			(useauxorigin no)
			(hpglpennumber 1)
			(hpglpenspeed 20)
			(hpglpendiameter 15.000000)
			(pdf_front_fp_property_popups yes)
			(pdf_back_fp_property_popups yes)
			(dxfpolygonmode yes)
			(dxfimperialunits yes)
			(dxfusepcbnewfont yes)
			(psnegative no)
			(psa4output no)
			(plotreference yes)
			(plotvalue yes)
			(plotfptext yes)
			(plotinvisibletext no)
			(sketchpadsonfab no)
			(subtractmaskfromsilk no)
			(outputformat 1)
			(mirror no)
			(drillshape 1)
			(scaleselection 1)
			(outputdirectory "")
		)
	)
	(net 0 "")
	(gr_rect
		(start 114 67)
		(end 148.5 93.5)
		(stroke
			(width 0.05)
			(type default)
		)
		(fill none)
		(layer "Edge.Cuts")
		(uuid "d0f407c0-0a76-440c-bef5-dda49c7c366e")
	)
	(zone
		(net 0)
		(net_name "")
		(layers "F&B.Cu")
		(uuid "80e3d796-596b-409e-b7e7-608af36e0685")
		(hatch edge 0.5)
		(connect_pads
			(clearance 0.5)
		)
		(min_thickness 0.25)
		(filled_areas_thickness no)
		(fill yes
			(thermal_gap 0.5)
			(thermal_bridge_width 0.5)
			(island_removal_mode 1)
			(island_area_min 10)
		)
		(polygon
			(pts
				(xy 110 60) (xy 155 60) (xy 155 100) (xy 110 100)
			)
		)
		(filled_polygon
			(layer "F.Cu")
			(island)
			(pts
				(xy 147.942539 67.520185) (xy 147.988294 67.572989) (xy 147.9995 67.6245) (xy 147.9995 92.8755)
				(xy 147.979815 92.942539) (xy 147.927011 92.988294) (xy 147.8755 92.9995) (xy 114.6245 92.9995)
				(xy 114.557461 92.979815) (xy 114.511706 92.927011) (xy 114.5005 92.8755) (xy 114.5005 67.6245)
				(xy 114.520185 67.557461) (xy 114.572989 67.511706) (xy 114.6245 67.5005) (xy 147.8755 67.5005)
			)
		)
		(filled_polygon
			(layer "B.Cu")
			(pts
				(xy 147.942539 67.520185) (xy 147.988294 67.572989) (xy 147.9995 67.6245) (xy 147.9995 92.8755)
				(xy 147.979815 92.942539) (xy 147.927011 92.988294) (xy 147.8755 92.9995) (xy 114.6245 92.9995)
				(xy 114.557461 92.979815) (xy 114.511706 92.927011) (xy 114.5005 92.8755) (xy 114.5005 67.6245)
				(xy 114.520185 67.557461) (xy 114.572989 67.511706) (xy 114.6245 67.5005) (xy 147.8755 67.5005)
			)
			(island)
		)
	)
)

```
</details>

ruben-iteng, 2025-01-05:
I think we should do the following:
https://github.com/atopile/atopile/blob/42a70810fe54943fefe5642b9ed4a947aedee511/src/faebryk/libs/kicad/fileformats.py#L937-L939

should be
```python
island: Optional[bool] = field(
    **sexp_field(positional=True), default=True
)
```

and a post init function for `class C_zone`
```python
def __post_init__(self):
    if self.net_name != "" and self.net != 0:
        for fpoly in self.filled_polygon:
            fpoly.island = False
```

as it seems a zone is only an island if it does not have a net.

Notes:
- This topic needs more research as KiCad might be buggy (does not always delete the island part of the zone)

mawildoer, 2025-01-10:
@ruben-iteng if you've tested this and are happy with the behaviour I think it makes as much sense as we'll be able to right now. We haven't run into this one ourselves yet as far as I can tell, but it's possible we saw something similar with `C_zone.enabled`

mawildoer, 2025-01-29:
I've managed to replicate this one and think it's actually a subtly different cause; it's a fault in the `CatchAll` and sexp de/encoder that's moving a positional `(island)` from slot 1 to the end.

I think there are three issues:

1. The positional `(island)` isn't being parsed properly
2. The decoder isn't properly rejecting additional statements without 
3. The `CatchAll` isn't re-placing values back in their correct indices

Addressing it now

PR #895 (2025-01-29): Fix `(island)` zones sexp decoding, `CatchAll` and `CatchAll` reproduction sequence
# Description

I've managed to replicate this one and think it's actually a subtly different cause; it's a fault in the `CatchAll` and sexp de/encoder that's moving a positional `(island)` from slot 1 to the end.

I think there are three issues:

1. The positional `(island)` isn't being parsed properly
2. The decoder isn't properly rejecting additional statements without 
3. The `CatchAll` isn't re-placing values back in their correct indices

Fixes #776 

# Checklist

Please read and execute the following:

-  [... cut]
commit f6088430 (2025-01-29): Fix `(island)` zones sexp decoding, `CatchAll` and `CatchAll` reproduction sequence (#895)

# Description

I've managed to replicate this one and think it's actually a subtly
different cause; it's a fault in the `CatchAll` and sexp de/encoder
that's moving a positional `(island)` from slot 1 to the end.

I think there are three issues:

1. The positional `(island)` isn't being parsed properly
2. The decoder isn't properly rejecting additional statements without 
3. The `CatchAll` isn't re-placing values back in their correct indices

Fixes #776
closed 2025-01-29 by mawildoer