atopile/atopile #778: `ato create project` fails when option no repo is choosen

ruben-iteng, 2025-01-02:
### Current Behaviour

`? Would you like to create a new repo for this project? No`
results in running a git command without proper config (= error)

### Expected Behaviour

The program exited with all "ok"

### Possible Solution

#pr number

### Version

0.3.4-dev

### Relevant log output

```shell
╭──────────────────────── Traceback (most recent call last) ────────────────────────╮
                    │ ..../atopile/src/atopile/cli/create.py:19 │
                    │ 4 in project                                                                      │
                    │                                                                                   │
                    │   191 │   # modifying things in gitignored locations                              │
                    │   192 │   if repo_obj.is_dirty():                                                 │
                    │   193 │   │   repo_obj.git.add(A=True, f=True)                                    │
                    │ ❱ 194 │   │   repo_obj.git.commit(m="Configure project")
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

mawildoer, 2025-01-31:
Tested this after #902 's updates today and couldn't replicate it, so I thiiiink it's fixed. Please reopen if it's still an issue!

ruben-iteng, 2025-02-01:
When choosing option `no`:

```
Would you like to create a new repo for this project? No
[23:00:07] ERROR    Uncaught compiler exception 
```
```
GitCommandError: Cmd('git') failed due to: exit code(128)                                                
  cmdline: git commit -m Configure project                                                               
  stderr: 'Author identity unknown                                                                       
                                                                                                         
*** Please tell me who you are.                                                                          
                                                                                                         
Run                                                                                                      
                                                                                                         
  git config --global user.email "you@example.com"                                                       
  git config --global user.name "Your Name"                                                              
                                                                                                         
to set your account's default identity.                                                                  
Omit --global to set the identity only in this repository.                                               
                                                                                                         
fatal: unable to auto-detect email address (got '<redacted>@<redacted>.(none)')'
```