atopile/atopile #807: Fails to resolve bus parameters

napowderly, 2025-01-10:
### Current Behaviour

F.is_bus_parameter.resolve_bus_parameters(G) fails occasionally
[image]

### Expected Behaviour

Retry automatically or not fail in the first place

### Possible Solution

#pr number

### Version

0.3.8.dev0

### Relevant log output

```shell
[10:41:02] ERROR    Exception                                                              
                    Unfortunately, there's a compiler bug at the moment that means that    
                    this sometimes fails. Try again, and it'll probably work.
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

napowderly, 2025-01-10:
This is typically found in larger designs, increasing the environment variables listed in the command below will solve the issue in some cases.
FBRK_MAX_PATHS=1e7 FBRK_MAX_PATHS_NO_WEAK=1e6 FBRK_MAX_PATHS_NO_NEW_WEAK=1e5 ato build -b default

iopapamanoglou, 2025-01-20:
duplicate #689

mawildoer, 2025-02-10:
[image]

Likely path discovery does not find all paths.
Can be fixed with higher path limit heuristics, but costs extra execution time.
Likely fixed in https://github.com/atopile/atopile/pull/523

napowderly, 2025-03-11:
Resolved!

closed 2025-03-11 by napowderly