atopile/atopile #814: Upgrade 0.2.x -> 0.3.x: Failed to decode kicad_pcb.layers.<C_layer>.type (<enum 'E_type'>)

hexdae, 2025-01-10:
### Current Behaviour

When building a minimal, no dependencies ato 0.2.0 project, auto 0.3.0 fails with the following error the existing layout project

```
           WARNING  pcbnew restart required (updated fp-lib-table)                                                                                                               
           ERROR    Decode Error                                                                                                                                                 
                    Failed to decode kicad_pcb.layers.<C_layer>.type (<enum 'E_type'>) with mixed   
```

### Expected Behaviour

The program exited with all checks "ok"

### Version

0.3.0

### Relevant log output

```shell
[17:07:01] INFO     Current version (0.3.8) is newer than latest (0.2.69)                                                                                                        
           INFO     Using project /DRV8833PW                                                                                         
           INFO     Building 'default'                                                                                                                                           
           WARNING  Exception                                                                                                                                                    
                    Ignoring assignment of http://www.ti.com/lit/ds/symlink/drv8833.pdf to datasheet on ic                                                                       
                    Source: elec/src/build.ato:3:7                                                                                                                               
                      2                                                                                                                                                          
                    ❱ 3 module Build:                                                                                                                                            
                      4 │   ic = new DRV8833PW                                                                                                                                   
                    Source: elec/src/build.ato:4:9                                                                                                                               
                      3 module Build:                                                                                                                                            
                    ❱ 4 │   ic = new DRV8833PW                                                                                                                                   
                      5                                                                                                                                                          
                    Source: DRV8833PW.ato:1:10                                                                                                                                   
                    ❱ 1 component DRV8833PW:                                                                                                                                     
                      2 │   datasheet = "http://www.ti.com/lit/ds/symlink/drv8833.pdf"                                                                                           
                    Code causing the error:                                                                                                                                      
                    Source: DRV8833PW.ato:2:4                                                                                                                                    
                      1 component DRV8833PW:                                                                                                                                     
                    ❱ 2 │   datasheet = "http://www.ti.com/lit/ds/symlink/drv8833.pdf"                                                                                           
                      3 │   description = "Motor / Motion / Ignition Controllers & Drivers DRV8833 EVAL MOD"                                                                     
           WARNING  Exception                                                                                                                                                    
                    Ignoring assignment of Motor / Motion / Ignition Controllers & Drivers DRV8833 EVAL MOD to description on ic                                                 
                    Source: elec/src/build.ato:3:7                                                                                                                               
                      2                                                                                                                                                          
                    ❱ 3 module Build:                                                                                                                                            
                      4 │   ic = new DRV8833PW                                                                                                                                   
                    Source: elec/src/build.ato:4:9                                                                                                                               
                      3 module Build:                                                                                                                                            
                    ❱ 4 │   ic = new DRV8833PW                                                                                                                                   
                      5                                                                                                                                                          
                    Source: DRV8833PW.ato:1:10                                                                                                                                   
                    ❱ 1 component DRV8833PW:                                                                                                                                     
                      2 │   datasheet = "http://www.ti.com/lit/ds/symlink/drv8833.pdf"                                                                                           
                    Code causing the error:                                                                                                                                      
                    Source: DRV8833PW.ato:3:4                                                                                                                                    
                      2 │   datasheet = "http://www.ti.com/lit/ds/symlink/drv8833.pdf"                                                                                           
                    ❱ 3 │   description = "Motor / Motion / Ignition Controllers & Drivers DRV8833 EVAL MOD"                                                                     
                      4 │   footprint = "SOP65P640X120-16N"                                                                                                                      
           INFO     Resolving bus parameters                                                                                                                                     
           INFO     Running checks                                                                                                                                               
           INFO     Checking graph for ERC violations                                                                                                                            
           INFO     Checking 0 Power                                                                                                                                             
[... cut ...]
[17:07:02] ERROR    Decode Error                                                                                                                                                 
                    Failed to decode kicad_pcb.layers.<C_layer>.type (<enum 'E_type'>) with mixed                                                                                

Unfortunately errors ^^^ stopped the build. If you need a hand jump on Discord! https://discord.gg/mjtxARsr9V 👋
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

mawildoer, 2025-01-11:
Looks like KiCAD supports more layer types than we do here: https://github.com/atopile/atopile/blob/93faf6695761be0827902ad5826a1d2a313ce467/src/faebryk/libs/kicad/fileformats.py#L728-L730

Expanding this should fix the issue! Thanks @hexdae !! 🙌

hexdae, 2025-01-11:
Expanded to follow the official [kicad docs ](https://dev-docs.kicad.org/en/file-formats/sexpr-pcb/index.html#_layers_section)

closed 2025-01-12 by mawildoer