atopile/atopile #815: Upgrade 0.2.x -> 0.3.x: footprints import broken, fp-lib-info issue

hexdae, 2025-01-10:
### Current Behaviour

Even after working around #814, when running `ato build` on this project that used to build with 0.2.0, the footprints and 3D models are not synced.

### Expected Behaviour

Layout is updated with the correct 3D models and footprint

### Possible Solution

N/A

### Version

0.3.x

### Relevant log output

```shell
[17:16:33] INFO     Current version (0.3.8) is newer than latest (0.2.69)                                                                                                        
           INFO     Using project /Users/davide/src/diodeinc/stdlib/components/DRV8833PW                                                                                         
           INFO     Building 'default'                                                                                                                                           
           WARNING  Exception                                                                                                                                                    
                    Ignoring assignment of http://www.ti.com/lit/ds/symlink/drv8833.pdf to datasheet on ic                                                                       
                    Source: elec/src/build.ato:3:7                                                                                                                               
                      2                                                                                                                                                          
                    ❱ 3 module Build:                                                                                                                                            
                      4 │   ic = new DRV8833PW                                                                                                                                   
                    Source: elec/src/build.ato:4:9                                                                                                                               
                      3 module Build:                                                                                                                                            
                    ❱ 4 │   ic = new DRV8833PW                                                                                                                                   
                      5                                                                                                                                                          
                    Source: DRV8833PW.ato:1:10                                                                                                                                   
                    ❱ 1 component DRV8833PW:                                                                                                                                     
                      2 │   datasheet = "http://www.ti.com/lit/ds/symlink/drv8833.pdf"                                                                                           
                    Code causing the error:                                                                                                                                      
                    Source: DRV8833PW.ato:2:4                                                                                                                                    
                      1 component DRV8833PW:                                                                                                                                     
                    ❱ 2 │   datasheet = "http://www.ti.com/lit/ds/symlink/drv8833.pdf"                                                                                           
                      3 │   description = "Motor / Motion / Ignition Controllers & Drivers DRV8833 EVAL MOD"                                                                     
           WARNING  Exception                                                                                                                                                    
                    Ignoring assignment of Motor / Motion / Ignition Controllers & Drivers DRV8833 EVAL MOD to description on ic                                                 
                    Source: elec/src/build.ato:3:7                                                                                                                               
                      2                                                                                                                                                          
                    ❱ 3 module Build:                                                                                                                                            
                      4 │   ic = new DRV8833PW                                                                                                                                   
                    Source: elec/src/build.ato:4:9                                                                                                                               
                      3 module Build:                                                                                                                                            
                    ❱ 4 │   ic = new DRV8833PW                                                                                                                                   
                      5                                                                                                                                                          
                    Source: DRV8833PW.ato:1:10                                                                                                                                   
                    ❱ 1 component DRV8833PW:                                                                                                                                     
                      2 │   datasheet = "http://www.ti.com/lit/ds/symlink/drv8833.pdf"                                                                                           
                    Code causing the error:                                                                                                                                      
                    Source: DRV8833PW.ato:3:4                                                                                                                                    
                      2 │   datasheet = "http://www.ti.com/lit/ds/symlink/drv8833.pdf"                                                                                           
                    ❱ 3 │   description = "Motor / Motion / Ignition Controllers & Drivers DRV8833 EVAL MOD"                                                                     
                      4 │   footprint = "SOP65P640X120-16N"                                                                                                                      
           INFO     Resolving bus parameters                                                                                                                                     
           INFO     Running checks                                                                                                                                               
           INFO     Checking graph for ERC violations                                                                                                                            
           INFO     Checking 0 Power                                                                                                                                             
           INFO     Checking 0 explicit nets                                                                                                                                     
           INFO     Checking 0 passives                                                                                                                                          
[... cut ...]
           INFO     Building 'variable-report' for 'default' config                                                                                                              
           INFO     Writing parameters to /Users/davide/src/diodeinc/stdlib/components/DRV8833PW/build/default.variables.md                                                      
           INFO     Built 'manifest', 'layout-module-map', 'bom', 'variable-report' for 'default' config                                                                         
           INFO     Build successful! 🚀
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

mawildoer, 2025-01-11:
The footprints should be fixed in #798, via `create_footprint_library`. The 3D models require similar attention

mawildoer, 2025-01-27:
@hexdae was this fixed by #813 ?

hexdae, 2025-01-27:
@mawildoer, seems like the build now just removes all custom footprints from the layout file, so not fixed for us yet.

But now the updated files are correct, definitely a step in the right direction. Zip below

[component-empty.zip](https://github.com/user-attachments/files/18565224/component-empty.zip)

hexdae, 2025-01-28:
I take this back, if I add some generics, the project can actually build and modify the layout, but providing custom footprints is still not working and the picker defaults to JLC