atopile/atopile #838: Relinking replaced component fails

napowderly, 2025-01-16:
From main:
[image]

napowderly, 2025-01-17:
Seems like when we specialize, we change the address of the component which causes the relinking to fail.

mawildoer, 2025-01-17:
This is actually a bit of a thorny problem.

In v0.2, we only replace a component's type. It's otherwise the same component, which means that we needn't worry about things like the address matching etc... because of course it does!

In v0.3 there's a specific relationship of specialization. This is vastly more powerful because, for example, it allows you to specialize to one op-amp within a package containing four, but is also thorny because we need to specially handle things like:
- What address should a specialised module have?
- Which traits (if any) should the specialised module inherit?

PR #844 (2025-01-20): Handle specialization in `ato` the same as v0.2
This relinks v0.2 designs neatly, but carries some tech debt, as mentioned in https://github.com/atopile/atopile/issues/838#issuecomment-2599315558

This needs to be resolved more thoroughly mid-term, since this won't work (at least well) for `fabll` modules, but I see no clean way outta it right now.

Fixes: #838
commit a7b74fd4 (2025-01-20): Handle specialization in `ato` the same as v0.2 (#844)

This relinks v0.2 designs neatly, but carries some tech debt, as
mentioned in
https://github.com/atopile/atopile/issues/838#issuecomment-2599315558

This needs to be resolved more thoroughly mid-term, since this won't
work (at least well) for `fabll` modules, but I see no clean way outta
it right now.

Fixes: #838
closed 2025-01-20 by mawildoer