atopile/atopile #843: Designator prefix and `F.has_designator_prefix` doesn't work

mawildoer, 2025-01-17:
Everything new ends up with `Uxxx`

iopapamanoglou, 2025-01-20:
Can't reproduce

```python
import faebryk.library._F as F
from faebryk.core.module import Module

class App(Module):
    battery: F.Battery

    def __preinit__(self) -> None:
        self.battery.add(F.has_designator_prefix("BAR"))
        self.battery.add(
            F.has_explicit_part.by_supplier(
                "C5239862",
                pinmap={
                    "1": self.battery.power.lv,
                    "2": self.battery.power.hv,
                },
            )
        )

```

[image]

iopapamanoglou, 2025-01-20:
(Probably) unrelated but def useful:
```diff
--- a/src/faebryk/libs/app/designators.py
+++ b/src/faebryk/libs/app/designators.py
@@ -58,8 +58,8 @@ def attach_random_designators(graph: Graph):
         if not n.has_trait(F.has_designator_prefix):
             prefix = type(n).__name__
             logger.warning(f"Node {prefix} has no designator prefix")
-
-        prefix = n.get_trait(F.has_designator_prefix).get_prefix()
+        else:
+            prefix = n.get_trait(F.has_designator_prefix).get_prefix()

```

ruben-iteng, 2025-02-04:
Can reproduce in my ato project (not in fabll).