atopile/atopile #913: Consolidate `generics` and standard library

mawildoer, 2025-01-31:
There are multiple standard libraries flying around:
- [`generics`](https://github.com/atopile/generics)
- [`F.`](https://github.com/atopile/atopile/tree/main/src/faebryk/library)

There are a few issues here:
- `generics` today are shimmed via `atopile.attributes`, which means it's unclear what you should look at and functionality is inconsistent between `fabll` and `ato` code.
- Upgrading generics is rough because there's no mechanism in old projects to get a version that they support
- The standard library contains more than it should, and some modules should be broken out into independent packages.

ruben-iteng, 2025-02-01:
Also the fabll library does not (and currently can not) contain pcb layouts from kicad files :(

mawildoer, 2025-02-02:
> Also the fabll library does not (and currently can not) contain pcb layouts from kicad files :(

That's actually not quite true! Check out https://github.com/atopile/atopile/blob/main/src/faebryk/library/has_reference_layout.py

ruben-iteng, 2025-02-02:
> > Also the fabll library does not (and currently can not) contain pcb layouts from kicad files :(
> 
> That's actually not quite true! Check out https://github.com/atopile/atopile/blob/main/src/faebryk/library/has_reference_layout.py

Oh cool, missed that one 😅 thx

closed 2025-02-04 by mawildoer