atopile/atopile #928: Picker: Handle bad/missing data components

iopapamanoglou, 2025-02-03:
### Feature Request

Example: `C98349`
- Inductor that misses attributes
- Used in esp32-s3

Behavior: Project can't pick because:
```
[11:44:25] ERROR    Pick Error                                                                                                                                                                                                                                       
                    No parts found that are compatible with design for _3v3_reg.inductor<inductance: ([1.2µH, 1.8µH]) rated_current: ([0A, 2.4A]) max_current: ([0A, infA]) saturation_current: ([3.5A]) self_resonant_frequency: ([0Hz, infHz]) dc_resistance: ([0Ω,
                    infΩ]) dcr: ([57.6mΩ, 86.4mΩ])>:                                                                                                                                                                                                                 
                                                                                                                                                                                                                                                                     
                     • C98349 (attributes: )  
```

Core issue: Bad/Missing data in our component DB
Workaround: Let user manually specify component (not module!!) data
Can't use normal parameters because ambiguous in this case.

Approach 1: Create local override db (file) where centrally all part parameter overrides are managed
Approach 2: Add trait (& for ato some syntax) to denote overrides for a part (instead of module)

I suggest approach 1

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

mawildoer, 2025-02-03:
This specification must be easy from ato code, and it must work with existing code to rebuild projects with v0.3. See: https://github.com/atopile/atopile/discussions/912

We maybe also could use ‘=‘ as “alias_is” instead of “constraint_subset” in ato component definitions. I don’t like the implicit behaviour, but otherwise I think it actually works out perhaps best

mawildoer, 2025-02-05:
Fixed in #933

closed 2025-02-05 by mawildoer