atopile/atopile #938: Connector footprints from easyeda -> kicad missing holes

napowderly, 2025-02-04:
### Current Behaviour

For example on usbc connector C709357:
[image]

Seems like both holes for the pins end up stacked on one side.

### Expected Behaviour

The program exited with all checks "ok"

### Possible Solution

#pr number

### Version

0.3.8

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

mawildoer, 2025-02-05:
This looks like it might be the root cause of #923

ruben-iteng, 2025-02-06:
> This looks like it might be the root cause of #923 
> 
> ( @ruben-iteng for vis)

Indeed same issue

ruben-iteng, 2025-02-21:
Looking at this issue again, it is the other way around: issue #923 is the cause of this issue (#938).

The atopile KiCad plug-in is probably doing something with the footprints.
Module KiCad layout is ok, top level layout with synced layouts has this issue.