atopile/atopile #1265: Part lifecycle manager: 3D file is not included

ruben-iteng, 2025-06-08:
### Current Behaviour

When no build dir and no part dir is present, then building your project (and thus downloading/creating the part lib etc, the generated parts will bit include the step file (in parts/<partname>/) but the step file is in the build/cache/parts/<lcscid>/ folder.

Reproduce by:
1. deleting build and parts folder
2. build project
3. …no 3D files in pars folder, but they are in the build/cache/.. folder
4. delete parts folder
5. build project
6. now there are 3D files in the parts folder

### Expected Behaviour

3D files are included in the parts folder

### Possible Solution

-

### Version

0.9.0

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

iopapamanoglou, 2025-06-10:
Slightly related:

When making PCB for first time and exporting using kicad cli, no 3d models.
Saving pcb with pcbnew and rerunning works

iopapamanoglou, 2025-06-17:
Fixed in #1283

closed 2025-06-17 by iopapamanoglou